tobozo/M5Stack-SD-Updater #3: M5Stack-SD-Menu compilation error

reaper7, 2018-03-19:
I try compile new example M5Stack-SD-Menu
but I got errors:
```
In file included from C:\PROGRAMY\arduino_projekty\libraries\M5Stack-SD-Updater\examples\M5Stack-SD-Menu\M5Stack-SD-Menu.ino:75:0:

sketch\joyPSP.h: In function 'void handleJoyPad()':

joyPSP.h:76:14: error: 'class M5SAM' has no member named 'setListID'

       M5Menu.setListID(MenuID);

              ^

joyPSP.h:90:14: error: 'class M5SAM' has no member named 'setListID'

       M5Menu.setListID(MenuID);

              ^

C:\PROGRAMY\arduino_projekty\libraries\M5Stack-SD-Updater\examples\M5Stack-SD-Menu\M5Stack-SD-Menu.ino: In function 'void renderIcon(uint16_t)':

M5Stack-SD-Menu:191:30: error: cannot convert 'FileInfo' to 'uint16_t {aka short unsigned int}' for argument '1' to 'void renderIcon(uint16_t)'

   renderIcon(fileInfo[MenuID]);

                              ^

exit status 1
'class M5SAM' has no member named 'setListID'
```

when I comment out //#define USE_PSP_JOY true
then got:
```
C:\tmp\arduino_modified_sketch_208768\M5Stack-SD-Menu.ino: In function 'void renderIcon(uint16_t)':

M5Stack-SD-Menu:191:30: error: cannot convert 'FileInfo' to 'uint16_t {aka short unsigned int}' for argument '1' to 'void renderIcon(uint16_t)'

   renderIcon(fileInfo[MenuID]);

                              ^

exit status 1
cannot convert 'FileInfo' to 'uint16_t {aka short unsigned int}' for argument '1' to 'void renderIcon(uint16_t)'
```

tobozo, 2018-03-20:
whoops I totally forgot I was using a modified M5Sam library, also forgot to PR the changes or even document it ... shame on me 👎 

[there's the pull request](https://github.com/tomsuch/M5StackSAM/pull/2)

tobozo, 2018-03-20:
I'm not sure why you got the other error, I can't reproduce it on a fresh environment, could it have something to do with the version of the compiler you're using?

It's probably confused because I'm using the same name for a global and a scoped variable.
You'll get the confirmation by changing the name of the scoped variable, here I renamed "fileInfo" to "myFileInfo" :

```
/* by file info */
void renderIcon(FileInfo &myFileInfo) {
  if(!myFileInfo.hasMeta || !myFileInfo.hasIcon) {
    return;
  }
  JSONMeta jsonMeta = myFileInfo.jsonMeta;
  M5.Lcd.drawJpgFile(SD, myFileInfo.iconName.c_str(), M5.Lcd.width()-jsonMeta.width-10, (M5.Lcd.height()/2)-(jsonMeta.height/2)+10, jsonMeta.width, jsonMeta.height, 0, 0, JPEG_DIV_NONE);
}
```

reaper7, 2018-03-20:
win10 64; arduino ide 1.9.0b; arduino-esp32 current git
both at home and at work the same error:
```
C:\Temp\arduino_modified_sketch_873693\M5Stack-SD-Menu.ino: In function 'void renderIcon(uint16_t)':

M5Stack-SD-Menu:191:30: error: cannot convert 'FileInfo' to 'uint16_t {aka short unsigned int}' for argument '1' to 'void renderIcon(uint16_t)'

   renderIcon(fileInfo[MenuID]);
```
rename not change anything, still the same.

I also try change the functions order:
renderIcon (by file) before renderIcon (by ID) 
or declare **void renderIcon(FileInfo &myFileInfo);** near line 102
then:
```

M5Stack-SD-Menu:163:34: error: 'lastcsize' may be used uninitialized in this function [-Werror=maybe-uninitialized]

   int16_t voffset = scrollPointer%lastcsize;

exit status 1
'lastcsize' may be used uninitialized in this function [-Werror=maybe-uninitialized]
```

reaper7, 2018-03-20:
strange...
gcc compiler for esp32 is more restrictive than esp8266
for e.g. :
QRCode example on esp8266 give tons of **warnings** for #pragma:
```
C:\PROGRAMY\arduino_projekty\libraries\QRCode\src\qrcode.c:37:0: warning: ignoring #pragma mark  [-Wunknown-pragmas]

 #pragma mark - Error Correction Lookup tables
```
QRCode example on esp32 give tons of **errors** for the same #pragma and break compilation!:
```
C:\PROGRAMY\arduino_projekty\libraries\QRCode\src\qrcode.c:37:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]

 #pragma mark - Error Correction Lookup tables
```

tobozo, 2018-03-20:
Are you on a Mac ? I stil can't reproduce this either on Windows 10 or Linux (Ubuntu 16) so I'm wondering if this is system specific.

I'm using xtensa-esp32-elf-gcc\1.22.0-61-gab8375a-5.2.0 on Windows 10, I'll check the Ubuntu version later when back home.

I know the M5Sam library gives me tons of warnings too about the TwoWires library, but those are only warnings and can be ignored as long as the sketch works.

Maybe your installation of ESP32 SDK is older than mine?

I update it once a week but sometimes forget to run `git submodule update` and run the `tools/get.py` utility,

reaper7, 2018-03-20:
windows10 both at home and at work

submodules updated and tools too of course

looks like we have different gcc 1.22.0-61 vs 1.22.0-80

```
$ ./get.exe
System: Windows, Info: Windows-10-10.0.16299
Platform: i686-mingw32
Tool xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip already downloaded
Extracting xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
Tool esptool-4dab24e-windows.zip already downloaded
Extracting esptool-4dab24e-windows.zip
Tool mkspiffs-0.2.2-arduino-esp32-win32.zip already downloaded
Extracting mkspiffs-0.2.2-arduino-esp32-win32.zip
Renaming mkspiffs-0.2.2-arduino-esp32-win32/ to mkspiffs
Done

```

tobozo, 2018-03-20:
that's the same version, OS and SDK

```
~\Documents\Arduino\hardware\espressif\esp32\tools [master ≡ +3 ~0 -0 !]> .\get.exe
System: Windows, Info: Windows-10-10.0.16299
Platform: i686-mingw32
Tool xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip already downloaded
```

I got the previous version number from the compile window in Arduino IDE, found out there's another ArduinoData/packages/esp32/tools folder, and removed the older GCC from there, but it still compiles fine with the latest version.

let's compare library versions, this is my stack:

```

Several libraries found for "WiFi.h"
Used : C:\Users\tobozo\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
Not Used : C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\libraries\WiFi
Several libraries found for "SD.h"
Used : C:\Users\tobozo\Documents\Arduino\hardware\espressif\esp32\libraries\SD
Not Used : C:\Users\tobozo\Documents\Arduino\libraries\SD
Not Used : C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\libraries\SD
Using Library M5Stack version 0.1.6 in folder: C:\Users\tobozo\Documents\Arduino\libraries\M5Stack 
Using Library WiFi version 1.0 in folder: C:\Users\tobozo\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi 
Using Library Wire version 1.0 in folder: C:\Users\tobozo\Documents\Arduino\hardware\espressif\esp32\libraries\Wire 
Using Library SPI version 1.0 in folder: C:\Users\tobozo\Documents\Arduino\hardware\espressif\esp32\libraries\SPI 
Using Library FS version 1.0 in folder: C:\Users\tobozo\Documents\Arduino\hardware\espressif\esp32\libraries\FS 
Using Library SD version 1.0.5 in folder: C:\Users\tobozo\Documents\Arduino\hardware\espressif\esp32\libraries\SD 
Using Library M5Stack-SD-Updater version 0.0.1 in folder: C:\Users\tobozo\Documents\Arduino\libraries\M5Stack-SD-Updater 
Using Library Update version 1.0 in folder: C:\Users\tobozo\Documents\Arduino\hardware\espressif\esp32\libraries\Update 
Using Library M5StackSAM version 0.0.2 in folder: C:\Users\tobozo\Documents\Arduino\libraries\M5StackSAM 
Using Library ArduinoJson version 5.13.1 in folder: C:\Users\tobozo\Documents\Arduino\libraries\ArduinoJson 
Using Library QRCode version 0.0.1 in folder: C:\Users\tobozo\Documents\Arduino\libraries\QRCode 
```

reaper7, 2018-03-20:
the same...
```
Znaleziono wiele bibliotek w "WiFi.h"
Wykorzystane: C:\PROGRAMY\arduino\hardware\espressif\esp32\libraries\WiFi
Niewykorzystane: C:\PROGRAMY\arduino\libraries\WiFi
Znaleziono wiele bibliotek w "SD.h"
Wykorzystane: C:\PROGRAMY\arduino\hardware\espressif\esp32\libraries\SD
Niewykorzystane: C:\PROGRAMY\arduino\libraries\SD
Użycie biblioteki M5Stack w wersji 0.1.6 z folderu: C:\PROGRAMY\arduino_projekty\libraries\M5Stack 
Użycie biblioteki WiFi w wersji 1.0 z folderu: C:\PROGRAMY\arduino\hardware\espressif\esp32\libraries\WiFi 
Użycie biblioteki Wire w wersji 1.0 z folderu: C:\PROGRAMY\arduino\hardware\espressif\esp32\libraries\Wire 
Użycie biblioteki SPI w wersji 1.0 z folderu: C:\PROGRAMY\arduino\hardware\espressif\esp32\libraries\SPI 
Użycie biblioteki FS w wersji 1.0 z folderu: C:\PROGRAMY\arduino\hardware\espressif\esp32\libraries\FS 
Użycie biblioteki SD w wersji 1.0.5 z folderu: C:\PROGRAMY\arduino\hardware\espressif\esp32\libraries\SD 
Użycie biblioteki M5Stack-SD-Updater w wersji 0.0.1 z folderu: C:\PROGRAMY\arduino_projekty\libraries\M5Stack-SD-Updater 
Użycie biblioteki Update w wersji 1.0 z folderu: C:\PROGRAMY\arduino\hardware\espressif\esp32\libraries\Update 
Użycie biblioteki M5StackSAM w wersji 0.0.2 z folderu: C:\PROGRAMY\arduino_projekty\libraries\M5StackSAM 
Użycie biblioteki ArduinoJson w wersji 5.13.1 z folderu: C:\PROGRAMY\arduino_projekty\libraries\ArduinoJson 
Użycie biblioteki QRCode w wersji 0.0.1 z folderu: C:\PROGRAMY\arduino_projekty\libraries\QRCode 
```

tobozo, 2018-03-20:
- Not GCC
- Not library versions

However, all your libraries seem to be called from the Program folder instead of your user's Document folder.

... Arduino version maybe ? 
Mine shows *Arduino 1.8.5 (Windows Store 1.8.10.0)* in the title bar.

reaper7, 2018-03-20:
on 1.8.5 from arduino.cc site:
```
C:\Temp\arduino_modified_sketch_290359\M5Stack-SD-Menu.ino: In function 'void renderScroll(String&, uint8_t, uint8_t, uint16_t)':

C:\Temp\arduino_modified_sketch_290359\M5Stack-SD-Menu.ino:129:17: warning: unused variable 'now' [-Wunused-variable]

   unsigned long now = millis();

                 ^

M5Stack-SD-Menu:162: error: 'lastcsize' may be used uninitialized in this function [-Werror=maybe-uninitialized]

   int16_t voffset = scrollPointer%lastcsize;

                                  ^

cc1plus.exe: some warnings being treated as errors

Znaleziono wiele bibliotek w "WiFi.h"
Wykorzystane: C:\PROGRAMY\arduino185\hardware\espressif\esp32\libraries\WiFi
Niewykorzystane: C:\PROGRAMY\arduino185\libraries\WiFi
Znaleziono wiele bibliotek w "SD.h"
Wykorzystane: C:\PROGRAMY\arduino185\hardware\espressif\esp32\libraries\SD
Niewykorzystane: C:\PROGRAMY\arduino185\libraries\SD
Użycie biblioteki M5Stack w wersji 0.1.6 z folderu: C:\PROGRAMY\arduino_projekty\libraries\M5Stack 
Użycie biblioteki WiFi w wersji 1.0 z folderu: C:\PROGRAMY\arduino185\hardware\espressif\esp32\libraries\WiFi 
Użycie biblioteki Wire w wersji 1.0 z folderu: C:\PROGRAMY\arduino185\hardware\espressif\esp32\libraries\Wire 
Użycie biblioteki SPI w wersji 1.0 z folderu: C:\PROGRAMY\arduino185\hardware\espressif\esp32\libraries\SPI 
Użycie biblioteki FS w wersji 1.0 z folderu: C:\PROGRAMY\arduino185\hardware\espressif\esp32\libraries\FS 
Użycie biblioteki SD w wersji 1.0.5 z folderu: C:\PROGRAMY\arduino185\hardware\espressif\esp32\libraries\SD 
Użycie biblioteki M5Stack-SD-Updater w wersji 0.0.1 z folderu: C:\PROGRAMY\arduino_projekty\libraries\M5Stack-SD-Updater 
Użycie biblioteki Update w wersji 1.0 z folderu: C:\PROGRAMY\arduino185\hardware\espressif\esp32\libraries\Update 
Użycie biblioteki M5StackSAM w wersji 0.0.2 z folderu: C:\PROGRAMY\arduino_projekty\libraries\M5StackSAM 
Użycie biblioteki ArduinoJson w wersji 5.13.1 z folderu: C:\PROGRAMY\arduino_projekty\libraries\ArduinoJson 
Użycie biblioteki QRCode w wersji 0.0.1 z folderu: C:\PROGRAMY\arduino_projekty\libraries\QRCode 
exit status 1
'lastcsize' may be used uninitialized in this function [-Werror=maybe-uninitialized]
```

tobozo, 2018-03-20:
maybe it'll shut up with this initialization instead:
Line 137:

  ```uint8_t csize = 0, lastcsize = 0;```

tobozo, 2018-03-20:
ok I found a way to reproduce one of those warnings, 
please try File -> Preferences -> Compiler warnings: set it to Default and try to compile again

reaper7, 2018-03-20:
it solves errors problems with:
- #pragma in qrcode lib 
- uninitialized lastcsize (without changing line 137)

a bit strange...for what reason the warning settings functionality sometimes turns it into errors ?
<br>
but still occurs error:
```
cannot convert 'FileInfo' to 'uint16_t {aka short unsigned int}' for argument '1' to 'void renderIcon(uint16_t)'
```

Two options how to solve this problem (in my case):
1. swap functions order: renderIcon (by file) before renderIcon (by ID) 
or
2. declare void renderIcon(FileInfo &myFileInfo); around the line 103

without above, for unknown reasons, first function renderIcon (by ID) knows nothing about second function renderIcon (by file).

tobozo, 2018-03-20:
Whoops swapping functions order is the thing I should have done in the beginning.
Big thanks for pointing that out, now I know those are hinting at best practices, I'm re-enabling warnings everywhere ^^

I have a theory on those inconsistencies with the compiler though.
In my home directories I found two installations of the ESP32 SDK on both Windows 10 and Ubuntu. 

I suspect the install directories changed over time (moving from ~/.arduinoxxx to ~/Arduino or to the directory set in the preferences.txt), and multiple copies were made. Now the IDE does not know which one to use as all the SDK are visible.

If you look for the "esp32" directories on your hard drive, I bet you'll find more than one with a subdirectory "tools".

Although I didn't have the symptoms you had, I renamed the oldest "ESP32" folders to "ESP32.old" and the path to the compiler changed in the IDE.

Could it be that you had a separate /duplicate version of the SDK using different compiler flags.
The M5Stack board used to be in a separate environment before it was merged into the ESP32 boards list, maybe some dotfiles are still there ?

I've pushed the [changes to the repo anyway](https://github.com/tobozo/M5Stack-SD-Updater/commit/7c3cd8b7d4c65c39d784e560cb28cf5274fab0bd), I want to thank you again for finding those anomalies and taking the time to test it, very much appreciated!

reaper7, 2018-03-20:
I have only one esp32 dir, because I don't use arduino installators, boards managers etc
**C:/PROGRAMY/Arduino** is a unzipped copy of arduino ide from arduino.cc->Windows ZIP file for non admin install
**C:/PROGRAMY/Arduino/hardware/espressif/esp32** clone of arduino-esp32 git repository
**C:/ PROGRAMY/arduino_projekty** is a dir for my sketches (set in Arduino->Preference->Sketchbook location
**C:/ PROGRAMY/arduino_projekty/libraries** contains my collection of external libs (99% from github) updated by bash script.

but does not matter...
important that the problem is solved :D

closed 2018-03-20 by reaper7