tobozo/M5Stack-SD-Updater #5: Compilation Errors

Kabron287, 2018-03-28:
I use VisualMicro plugin (visualmicro.com). Previous version compiled without any problems.
With the new one I got:

  M5Stack-SD-Menu.ino: 80:35: error: 'JSONMeta' has not been declared
   void getMeta(String metaFileName, JSONMeta &jsonMeta)
M5Stack-SD-Menu.ino: 82:17: error: variable or field 'renderIcon' declared void
   void renderIcon(FileInfo &fileInfo)
M5Stack-SD-Menu.ino: 82:17: error: 'FileInfo' was not declared in this scope
M5Stack-SD-Menu.ino: 82:27: error: 'fileInfo' was not declared in this scope
   void renderIcon(FileInfo &fileInfo)
M5Stack-SD-Menu.ino: 84:17: error: variable or field 'renderMeta' declared void
   void renderMeta(JSONMeta &jsonMeta)
M5Stack-SD-Menu.ino: 84:17: error: 'JSONMeta' was not declared in this scope
M5Stack-SD-Menu.ino: 84:27: error: 'jsonMeta' was not declared in this scope
   void renderMeta(JSONMeta &jsonMeta)

In Arduino IDE another problem:

D:\Arduino\libraries\QRCode\src\qrcode.c:37:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - Error Correction Lookup tables
 ^
D:\Arduino\libraries\QRCode\src\qrcode.c:100:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - Mode testing and conversion
 ^
D:\Arduino\libraries\QRCode\src\qrcode.c:139:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - Counting
 ^
D:\Arduino\libraries\QRCode\src\qrcode.c:166:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - BitBucket
 ^
D:\Arduino\libraries\QRCode\src\qrcode.c:250:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - Drawing Patterns
 ^
D:\Arduino\libraries\QRCode\src\qrcode.c:473:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - Penalty Calculation
 ^
D:\Arduino\libraries\QRCode\src\qrcode.c:573:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - Reed-Solomon Generator
 ^
D:\Arduino\libraries\QRCode\src\qrcode.c:627:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - QrCode
 ^
D:\Arduino\libraries\QRCode\src\qrcode.c:768:0: error: ignoring #pragma mark  [-Werror=unknown-pragmas]
 #pragma mark - Public QRCode functions

tobozo, 2018-03-28:
Hello and thank you for your feedback

The first error `M5Stack-SD-Menu.ino: 80:35: error: 'JSONMeta' has not been declared` makes me wonder what version you're using, did you download & unzip manually or did you get the library from the library manager?

see #3 for reference

Some fixes have been applied recently, so if your version is earlier, it might be solved by updating it.

The other #pragma errors on Arduino can be solved by setting the compiler warnings to "default".

vMicro > Compiler Warnings menu item > see if you can see a "default" or "silent" value.

Kabron287, 2018-03-28:
Arduino IDE1.8.5, vMicro 1803.26.0
I manually downloaded and install. date of the lib is 28.03.2018 15:22 (GMT+3)

Kabron287, 2018-03-28:
I decreased warning level in Arduino IDE and now it compiled.

tobozo, 2018-03-28:
Pragma errors are raised by external libs so I can't do anything about it but I thought the other errors were solved by the latest update.

I'll update the ReadME since you're the second person to report this.

I'm closing this issue, thank you again for your feedback and don't hesitate to reopen it or create a new one if something weird comes up :+1:

tobozo, 2018-03-28:
Actually I'll reopen it and rename it as specific to vMicro since this part isn't solved.

ifrew, 2018-05-04:
Add these 3 lines before the existing definition of void getMeta in M5Stack-SD-Menu.ino and you are good to go! I'm running on visual micro 1711.19.0. Very neat code. Many thanks!

void getMeta(String metaFileName, JSONMeta &jsonMeta);
void renderIcon(FileInfo &fileInfo);
void renderMeta(JSONMeta &jsonMeta);

tobozo, 2018-05-08:
@ifrew thanks for the tip, will do that and probably do some namespacing too as I've spotted compilation problems with platformio too

tobozo, 2018-05-11:
@Kabron287 @ifrew thanks to you I've namespaced the library and added the signatures as suggested and I'll do a release soon. Can you confirm the latest clone of this repo does not break your compilation anymore ? If so I'll add vMicro to the list of supported environments.

ifrew, 2018-05-12:
Yip! Downloaded from the repository, deleted old library installed new one and rebuilt the example. All good now. Thanks.

tobozo, 2018-05-12:
fixed in [last release](https://github.com/tobozo/M5Stack-SD-Updater/releases/tag/v0.3.0), thanks a lot for you hel on this!

closed 2018-05-12 by tobozo

Kabron287, 2018-05-13:
At the moment, the only error:
M5Stack-SD-Menu.ino: 428:8: error: 'class M5SAM' has no member named 'setListID
M5Stack lib 0.1.8

tobozo, 2018-05-13:
What version of M5Stack-Sam are you using?
Make sure you use [this one](https://github.com/tomsuch/M5StackSAM)

Kabron287, 2018-05-13:
After updating M5StackSam compilation OK, but now I have permanent bootloop in debug and release.

tobozo, 2018-05-13:
Maybe update the M5Stack Core too ? Current version is [0.1.9](https://github.com/m5stack/M5Stack/releases/tag/0.1.9)

tobozo, 2018-05-13:
Do you get any warning during compilation?

Kabron287, 2018-05-13:
With M5Stack core 0.1.9
In_eSPI.cpp: 618:55: warning: 'reg' is used uninitialized in this function [-Wuninitialized]

M5StackSAM.cpp: 237:33: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second

joyFacesGameboy.h: 54:3: warning: 'r' may be used uninitialized in this function [-Wmaybe-uninitialized]

all others warning are about unused

Kabron287, 2018-05-13:
But to for saving compatibility with all my projects I have to add
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiMulti.h>
to M5stack.h
it does not affect before

tobozo, 2018-05-13:
I don't see any fatal problem here, did you restart your IDE after changing updating the libraries ? Maybe a linkage problem does this.

Of course if you modified the M5Stack Core, side effects can happen, are other sketches affected by this boot loop or does it only affect the SD Menu?

Kabron287, 2018-05-13:
Factory Test 0.1.9 works fine

tobozo, 2018-05-13:
add your includes to M5StackUpdater.h too then

Kabron287, 2018-05-13:
In Arduino IDE:
joyFacesGameboy.h:54: error: 'r' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   if (r == '4') { return UI_INFO; }

it is corresponding with Vmicro warning

tobozo, 2018-05-13:
are you using M5Stack faces? if not you can disable it in controls.h, you just have to comment out this line:

`#define USE_FACES_GAMEBOY`

Kabron287, 2018-05-13:
No progress :(

Kabron287, 2018-05-13:
In arduino IDE now:

C:\Arduino-ESP32-IDE\Proj\libraries\M5Stack\src\utility\In_eSPI.cpp:1199:15: error: value computed is not used [-Werror=unused-value]
         *ptr++;
               ^
C:\Arduino-ESP32-IDE\Proj\libraries\M5Stack\src\utility\In_eSPI.cpp:1348:15: error: value computed is not used [-Werror=unused-value]
         *ptr++;
               ^
C:\Arduino-ESP32-IDE\Proj\libraries\M5Stack\src\utility\In_eSPI.cpp:618:55: error: 'reg' is used uninitialized in this function [-Werror=uninitialized]
   reg |= (readcommand8(cmd_function, index + 0) <<  8);
                                                       ^
cc1plus.exe: some warnings being treated as errors
IDE\Proj\libraries\ArduinoJson 
exit status 1

tobozo, 2018-05-13:
yep back to the initial problem " cc1plus.exe: some warnings being treated as errors"

Anyway all this is more of a customized M5Stack problem than a library problem, especially since the last move to TFT_eSPI where things will stay unstable for a while.

So maybe stick will the versions you know it's working with until this is sorted out?

Kabron287, 2018-05-13:
I think so