tobozo/M5Stack-SD-Updater #13: Safe load from SD Card

valerio-vaccaro, 2018-04-26:
I really like this project, in my opinion, can be an interesting evolution sign all binaries with a private key and check using a public hardcoded in the firmware. 

In some weeks I will try to make it work in that way, someone interested in share more ideas for a safer implementation?

Ciao!

tobozo, 2018-04-26:
thanks for your feedback :-) 

signing binaries sounds much safer than the actual naive implementation 👍 but it also sounds like this would require ESP-IDF for flash encryption, right?

There may be an upcoming problem as existing hidden feature will be extended soon: currently any bin/jpg/mp3 file found on the SPIFFS is automatically moved to the microSD card.
This also needs to be implemented the other way around as some precompiled sketches will only work with their respective SPIFFS folder.

In an upcoming feature addition, SPIFFS content (when applicable) will be copied from a subfolder on the SD Card prior to loading a binary from the menu app, and backed up prior to unloading the app. 
Some apps are currently unusable/broken because of this lack of SPIFFS support.

Does it mean SPIFFS should be marked as secure? Also there's concern over files moved from/to the SD Card, should those be encrypted with a per-app key, then decrypted before being copied onto a SPIFFS secure partition, then re-encrypted when backed up on the SD?

Or maybe it's simpler if signed binaries don't use SPIFFS at all?

tobozo, 2018-05-14:
"dirty" load implemented: https://github.com/tobozo/M5Stack-SD-Updater/blob/master/examples/M5Stack-YOLO-Downloader/M5Stack-YOLO-Downloader.ino