tobozo/M5Stack-SD-Updater #16: error 'StaticJsonBuffer' was not declared in this scope

janseta, 2018-06-17:
Got this error with the Current library ArduinoJSON v 6.0.0-beta from the library manager in the Arduino IDE.

error: 'StaticJsonBuffer' was not declared in this scope

Downdating to the previous version ArduinoJSON version 5.13.2 solved it.

tobozo, 2018-06-17:
I have it on a few other projects too, I guess they messed up their releases flag and it came up in the update list even though it's beta.

tobozo, 2018-06-18:
This code will work with version 6x of ArduinoJSON:

```
void getMeta(String metaFileName, JSONMeta &jsonMeta) {
  File file = SD.open(metaFileName);
  StaticJsonDocument<512> jsonBuffer;
  DeserializationError error = deserializeJson(jsonBuffer, file);
  if (error) return;
  JsonObject &root = jsonBuffer.as<JsonObject>();
  
  if (root.success()) {
    jsonMeta.width  = root["width"];
    jsonMeta.height = root["height"];
    jsonMeta.authorName = root["authorName"].as<String>();
    jsonMeta.projectURL = root["projectURL"].as<String>();
    jsonMeta.credits    = root["credits"].as<String>();
  }
}

```

I'll create a new release soon with both codes

tobozo, 2018-07-23:
@janseta I updated the code on this repo so it works with any version of ArduinoJSON.

Can you confirm it compiles fine with your environment and close this issue if it does ?

eried, 2018-08-02:
I have the same error (Arduino 1.8.5, Windows 10). The new code does not fix it (`root` does not has `success()` error), but installing the older library fixes it, however the menu does not load anything. It is like that it tries to load (I see the progress bar) but then it reboots to the menu again. 

Could it be something with the SD card? I am using a small 512 MB one

tobozo, 2018-08-02:
@eried 
progress bar + reboots sounds like it's copying files from the SPIFFS to the SD, it that an infinite loop or does is stop after a while ?

It could also be a bad filesystem, missing or incomplete folder structure, missing menu.bin.

`root does not have success()` error is related to an early beta 6x version of ArduinoJSON, very unlikely someone will use it (unless doing exhaustive tests). 
You can safely update to the latest ArduinoJSON version, it has been tested successfully with the latest syntax.

This is my current stack (compiles fines, works fine, tested seconds ago) if you want to check for library version numbers:

```
Using library SPIFFS at version 1.0 in folder: ~/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/SPIFFS 
Using library FS at version 1.0 in folder: ~/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/FS 
Using library M5Stack at version 0.2.1 in folder: ~/Arduino/libraries/M5Stack 
Using library Wire at version 1.0 in folder: ~/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/Wire 
Using library SPI at version 1.0 in folder: ~/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/SPI 
Using library SD at version 1.0.5 in folder: ~/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/SD 
Using library M5Stack-SD-Updater at version 0.3.2 in folder: ~/Arduino/libraries/M5Stack-SD-Updater 
Using library Update at version 1.0 in folder: ~/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/Update 
Using library M5StackSAM at version 0.0.2 in folder: ~/Arduino/libraries/M5StackSAM 
Using library ArduinoJson at version 6.2.3-beta in folder: ~/Arduino/libraries/ArduinoJson 
```

As you can see most libraries are provided by the ESP32 SDK.

This leaves you with only M5Stack, M5Stack-SD-Updater, M5Stack-Sam and ArduinoJSON to check.

Also I'm using the latest ESP32 SDK, if you're still updating from the repo, now is time to checkout the release 1.0.0.

As an extra troubleshooting step you can also check the Serial output when the bootloop occurs.

tobozo, 2018-08-03:
`Error Occurred. Error #: 6` is an error message from Update.h

I suppose your M5 hasn't been flashed a million times, so it's probably the menu.bin got corrupted, or a previous OTA operation failed.

If you can format SPIFFS and erase_flash you'll be sure no OTA operation is pending (those may persist upon reboots).
The expected scenario after flashing a clean M5 is a SD scan + show file list, nothing else.

If you have a spare SD Card, it's worth re-compiling the menu.bin and copying it again, then compare.

eried, 2018-08-10:
I will try again tonight. But this is a new M5, so I guess it has't been flashed million times :)

tobozo, 2019-04-28:
this bug has now been fixed

closed 2019-04-28 by tobozo