tobozo/M5Stack-SD-Updater #17: BootLoop

Kabron287, 2018-07-23:
All components are the latest. Cmpilation via VisualMicro.
Result:
`Opening port
Port open
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400e3b99  PS      : 0x00060b34  A0      : 0x800e6c59  A1      : 0x3ffe7d20  
A2      : 0x3f4089b8  A3      : 0x000000bb  A4      : 0x3f408a14  A5      : 0x3f4088d8  
A6      : 0x00000005  A7      : 0x00000000  A8      : 0x800e3b99  A9      : 0x3ffe7cd0  
A10     : 0x400f7174  A11     : 0x3ffcc110  A12     : 0x3f40683c  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x400e3b99:0x3ffe7d20 0x400e6c56:0x3ffe7d50 0x400838b3:0x3ffe7d80 0x40083901:0x3ffe7da0 0x40007c31:0x3ffe7dc0 0x4000073d:0x3ffe7e30

Core 0 register dump:
PC      : 0x400e6a04  PS      : 0x00050034  A0      : 0x00000000  A1      : 0x3ffde620  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ff4808c  
A10     : 0x38004800  A11     : 0x00000000  A12     : 0x00060023  A13     : 0x3ffde6b4  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e6a04:0x3ffde620 0x7ffffffd:0x3ffde640

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
E (0) boot: Assert failed in esp_dport_access_int_init, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/dport_access.c:187 (res == pdTRUE)
`

tobozo, 2018-07-23:
Hey Vladimir,

Thank you for your feedback :-)

Can you [decode this exception](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html) from your environment and find where exactly the error occurs?

The last breaking change on this project was when ArduinoJSON jumped from 5.x to 6.x, and is has not been released yet (although it would give you compilation errors, not guru mediation), maybe it's worth cloning the lastest code of this repo and see if it improves ?

Kabron287, 2018-07-23:
Using IDF Monitor assumes to have ESP-IDF installed. I'm currently moving to a new PC with Win10 LTSB and have not yet installed all the programs. Under VisualMicro exception decoder for ESP32 is not working yet.

tobozo, 2018-07-23:
from the last line of the debug you pasted, it seems you have esp-idf installed.

```E (0) boot: Assert failed in esp_dport_access_int_init, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/dport_access.c:187 (res == pdTRUE)**```

Otherwise you can use the exception decoder from the Arduino IDE but you'll need to find the ELF file first and I have no idea how to do that from vsCode.

Kabron287, 2018-07-23:
**ficeto** is not my username. It is somebody from libraries developers

tobozo, 2018-07-23:
and I forgot to ask: are you compiling the SD-Menu app, or are you just compiling a personal sketch you want to load from the app?

Kabron287, 2018-07-23:
I'm compiling  SD-Menu app from the examples folder

tobozo, 2018-07-23:
if you know where the Arduino IDE drops the ELF file, you can decode the exception

Provided you're using the latest [ESP Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder) jar file, just run it and it'll ask you where the ELF file is, then paste your exception code there.

Kabron287, 2018-07-23:
Arduino gives or the first exception:
`0x400e3b99: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bootloader_support/src/bootloader_init.c line 523
0x400f7174: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c line 35
0x400e3b99: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bootloader_support/src/bootloader_init.c line 523
0x400e6c56: esp_dport_access_int_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/dport_access.c line 187 (discriminator 1)
0x400838b3: start_cpu1_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 406
0x40083901: call_start_cpu1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 245`

and for the second one:
`0x400e6a04: main_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 441 (discriminator 1)
0x400e6a04: main_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 441 (discriminator 1)`

tobozo, 2018-07-23:
Looks more like an environment problem to me, mostly because I don't see any reference to the sdUpdater code in the decoded exception.

Of course I can be wrong since I don't have any experience using vsCode, also I'm not that familiar with esp-idf.

Did you succeed into compiling & running other M5Stack sketches (which ones?) using the same environment?

Would vsCode let you capture the compilation log so I can inspect the library versions, sdk+tools path, and other interesting details?

Kabron287, 2018-07-23:
I guess you confuse vsCode and VisualMicro for VS. vsCode does not have adequate Arduino support(maybe except PlatformIO).
All my M5Stack sketches runs OK on M5Stack and ODROID-GO. I modified M5Stack 0.2.0 library to support both M5Stack and ODROID-GO.

Kabron287, 2018-07-23:
It is. Don't know about Linux. My Arduino IDE behaves the sama as VisualMicro.
Sure, after some testing.

tobozo, 2018-07-23:
nailed it !

edit the "controls.h" and comment out this line:

`//#define USE_FACES_GAMEBOY`

Kabron287, 2018-07-24:
Bingo! All works.
BTW it maybe interesting for you. 
SD load on ODROID-GO in games emulator mode works much faster. Maybe you find some usefull. It is opensource.

tobozo, 2018-07-24:
Aren't those games loaded into SPI Ram? That would explain the speed, also the SD Card type matters.

Actually I've been long speculating around an SD-Loader that would fit on any ESP32+SD+Display combination (ESP32-Wrover-Kit, Odroid-Go, M5, TTGO TS, ESP32-F-Dev-Kit).

The biggest issue being the use of M5Stack-Sam since it's hardwired to M5.Lcd, I'll need to replace that lib for every product.

Kabron287, 2018-07-24:
"The biggest issue being the use of M5Stack-Sam since it's hardwired to M5.Lcd"
It does not at all. Your SD-updater works on ODROID-GO fine. All hardware-related changes I did only in M5Stack lib.

tobozo, 2018-07-24:
Odroid-Go is built on a clone of M5Stack.h so it's no surprise M5Sam works there too.

I meant that it won't work out of the box on an ESP32-Wrover-Kit or on a custom build that uses a different resolution (such as [LoLin D32 Pro](https://www.aliexpress.com/item/LOLIN-D32-Pro-V2-0-0-wifi-bluetooth-board-based-ESP-32-esp32-Rev1-ESP32-WROVER/32883116057.html)).

Kabron287, 2018-07-24:
Because of using M5Stack lib for screenout. If to adapt M5Stack lib for other hardware it'll suit. In my plans spread it at least for Wrover kit as it's most similar for M5Stack and ODROID-GO.

timtimsson, 2018-09-23:
Hi,
i combile your ino and copy the hole file to SD...
But after i choise a binarie its load this and M5 goes in a boot loop?
Can you help?

```
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1496
load:0x40078000,len:8596
load:0x40080400,len:6980
entry 0x400806f4
M5Stack initializing...OK
Scanning SPIFFS for binaries
E (790) SPIFFS: mount failed, -10025
SPIFFS Mount Failed
Listing directory: /
  DIR : /System Volume Information
  FILE: /mp3-player.bin
  FILE: /MultiApps-Adv.bin
  FILE: /NyanCat.bin
  FILE: /NyanCat_Ext.bin
  FILE: /Oscilloscope.bin
  FILE: /PacketMonitor.bin
  FILE: /Pacman-JoyPSP.bin
  FILE: /Pixel-Fun.bin
  FILE: /Rickroll.bin
  FILE: /Sokoban.bin
  FILE: /SpaceDefense.bin
  FILE: /SpaceShooter.bin
  FILE: /Tetris.bin
  DIR : /jpg
  DIR : /json
  DIR : /mp3
  FILE: /arduinomegachess.bin
  FILE: /Colours_Demo.bin
  FILE: /CrackScreen.bin
  FILE: /d_invader.bin
  FILE: /drawNumber.bin
  FILE: /FlappyBird.bin
  IGNORED FILE: /menu.bin
  DIR : /data
  DIR : /mod
  FILE: /menu.bin
percent = 1
percent = 2
percent = 3
percent = 4
percent = 5
percent = 6
percent = 7
percent = 8
percent = 9
percent = 10
percent = 11
percent = 12
percent = 13
percent = 14
percent = 15
percent = 16
percent = 17
percent = 18
percent = 19
percent = 20
percent = 21
percent = 22
[... 182 lines cut ...]
load:0x40078000,len:8596
load:0x40080400,len:6980
entry 0x400806f4
assertion "false && "item should have been present in cache"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_item_hash_list.cpp", line 85, function: void nvs::HashList::erase(size_t)
abort() was called at PC 0x400df06b on core 0

Backtrace: 0x40087bc0:0x3ffc75e0 0x40087d9b:0x3ffc7600 0x400df06b:0x3ffc7620 0x400e22e8:0x3ffc7650 0x400e2656:0x3ffc7670 0x400e2985:0x3ffc76c0 0x400e20fc:0x3ffc7720 0x400e1f3e:0x3ffc7770 0x400e1fd7:0x3ffc7790 0x400e2022:0x3ffc77b0 0x400d6ea0:0x3ffc77d0 0x400f1613:0x3ffc77f0 0x400e08aa:0x3ffc7820

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1496
load:0x40078000,len:8596
load:0x40080400,len:6980
entry 0x400806f4
assertion "false && "item should have been present in cache"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_item_hash_list.cpp", line 85, function: void nvs::HashList::erase(size_t)
abort() was called at PC 0x400df06b on core 0

Backtrace: 0x40087bc0:0x3ffc75e0 0x40087d9b:0x3ffc7600 0x400df06b:0x3ffc7620 0x400e22e8:0x3ffc7650 0x400e2656:0x3ffc7670 0x400e2985:0x3ffc76c0 0x400e20fc:0x3ffc7720 0x400e1f3e:0x3ffc7770 0x400e1fd7:0x3ffc7790 0x400e2022:0x3ffc77b0 0x400d6ea0:0x3ffc77d0 0x400f1613:0x3ffc77f0 0x400e08aa:0x3ffc7820
```

tobozo, 2018-09-23:
hello @timtimsson 

This error:

`E (790) SPIFFS: mount failed, -10025
SPIFFS Mount Failed` 

.. makes me wonder if you compiled the binary with SPIFFS support. 

Maybe recompile with the default SPIFFS settings ? The requirements are the same as with OTA.

mega-byte, 2018-11-12:
I've got M5Stack up and running and successfully run several example  sketches; - wow what a great device! Next I  want to use the SD card and SD-Menu tools (I'm using the Arduino IDE) , and I'm struggling! For the benefit of others I'll detail some wider observations in the hope of helping others and filling in gaps in my understanding.  I am following the instructions at:
https://github.com/tobozo/m5stack-sd-updater 
Issues encountered being:
1. I did not know how to check my version of M5Stack but as I've only bought and started using it in the last month - ASSUME! its new enough?
2. What exactly is meant by "Open both sketches from the "examples/M5Stack-SD-Update" menu?  I have 3 sketches in File / Examples / M5Stack-SD-Updater namely M5Stack-SD-Menu, M5Stack-SDLoader-Snippet and M5Stack-YOLO-Downloader. I chose to merely open and work with the first of these.
3. I was able to compile, rename and copy the resulting bin to the SD card, then the Instructions  say 'Next, flash "menu.bin" to the M5Stack'. Not quite sure how to do this; firstly I ASSUME that to 'flash' we use the IDE upload technique, but hey, the IDE is still called M5Stack-SD-Menu - do I need to completely rename the folder and .ino to menu, or is there another dodge equivalent to "Upload as".?
4. I'm not 100% sure, however I think I've manged to get so far so good, as when I do upload the M5Stack-Sd-Menu sketch to the Arduino, it does give the Application Menu display, and I can scroll through the available apps using  Btn C. I also get the Scrolling link details and QR code displayd if I select info and canget back from that to the main menu.
5. Whenever I select any of the Applications however all I get is the loading progress bar up to 100% then nothing more. The serial monitor on Com 3 shows its in a loop trying to start something but failing each time - an extract is shown below.
17:36:21.319 -> Rebooting...
17:36:21.319 -> ets Jun  8 2016 00:22:57
17:36:21.319 -> 
17:36:21.319 -> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
17:36:21.319 -> configsip: 0, SPIWP:0xee
17:36:21.319 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:36:21.319 -> mode:DIO, clock div:1
17:36:21.319 -> load:0x3fff0018,len:4
17:36:21.319 -> load:0x3fff001c,len:1496
17:36:21.319 -> load:0x40078000,len:8596
17:36:21.319 -> load:0x40080400,len:6980
17:36:21.319 -> entry 0x400806f4
assertion "false && "item should have been present in cache"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_item_hash_list.cpp", line 85, function: void nvs::HashList::erase(size_t)
17:36:21.507 -> abort() was called at PC 0x400dec87 on core 0
17:36:21.507 -> 
17:36:21.507 -> Backtrace: 0x4008781c:0x3ffc7810 0x400879f7:0x3ffc7830 0x400dec87:0x3ffc7850 0x400e1f04:0x3ffc7880 0x400e2272:0x3ffc78a0 0x400e25a1:0x3ffc78f0 0x400e1d18:0x3ffc7950 0x400e1b5a:0x3ffc79a0 0x400e1bf3:0x3ffc79c0 0x400e1c3e:0x3ffc79e0 0x400d6950:0x3ffc7a00 0x400f122f:0x3ffc7a20 0x400e04c6:0x3ffc7a50
17:36:21.507 -> 
17:36:21.507 -> Rebooting...
17:36:21.507 ->

This looks identical to other reports in this thread however having seen this further up this :
nailed it !
edit the "controls.h" and comment out this line:
//#define USE_FACES_GAMEBOY
-I have checked my source and its already got that mod applied :-(      { or should it be :-) }

Another check I did was to try loading the example SPIFFS_Test sketch and that works ok.

I'm reaching the limit of my capability; Can anyone help, please?

tobozo, 2018-11-12:
@mega-byte most precompiled binaries provided in the current release probably won't work.
Both the SDK and the M5 library have been updated.recently use a different memory layout, which produces this kind of errors.

Maybe try with the binaries from this [more recent release](https://github.com/tobozo/M5Stack-SD-Updater/releases/download/untagged-a8c7518a642baec2980a/SD-Apps-Folder.zip) ?
These were compiled by [Travis](https://travis-ci.org/tobozo/M5Stack-SD-Updater) during last automatic build a couple of weeks ago.

mega-byte, 2018-11-12:
Hey tobozo thanks for the quick response. Unfortunately the link you provided to the binaries from a more recent release doesn't go anywhere and gives 'not found'. Also are you suggesting that that if I compile a new binary based on M5Stack-SDLoader-Snippet.ino  &/or recompile the supplied applications that this will get things rolling for me? Again any help most welcome......

tobozo, 2018-11-13:
Yep you need to compile new binaries to apply the new partition layout, here's the [fixed link](https://github.com/tobozo/M5Stack-SD-Updater/releases/download/v0.3.2/SD-Apps-Folder.zip) to the updated binaries.

mega-byte, 2018-11-13:
tobozo you are my hero! Now working. Some of the apps are not optimised for the M5Stack screen as delivered but hey ho; now I can use this framework to manage my own suite of stuff. Thanks again :-)

tobozo, 2019-04-28:
closing this as solved

closed 2019-04-28 by tobozo