tobozo/M5Stack-SD-Updater #18: some issues

grooves38, 2018-09-27:
Hi,
compiling, flashing works so far,
but when I enter an SD app I can't go back to menu.bin
I see, Loading menu.bin at LCD
and serial monitor shows a downloading unitl about 78%:
percent = 70
percent = 71
percent = 72
percent = 73
percent = 74
percent = 75
percent = 76
percent = 77
percent = 78
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

then a reboot occurs and previous App starts

the chess app ends up in an endless loop with:

st:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1496
load:0x40078000,len:8596
load:0x40080400,len:6980
entry 0x400806f4


Thanks for help
grooves

tobozo, 2018-09-27:
Hey and thanks for your feedback.

The chess App is the biggest of all apps, it may not work depending on your M5 model.

loading stops at 78% => this could be a SPIFFS partition problem i.e. too small or never used, maybe formatted with another SDK ?

Doing an full flash erase and some SPIFFS basic tests (e.g. upload the menu.bin and look what happens) could probably help.

grooves38, 2018-09-27:
thanks for your prompt answer,
I did already a full erase with esptool32.py erase_flash
after first upload with arduino IDE the menu.bin starts as expected,
I can select an app and can start the app
but as written before .. it reboots at 78% when I want to go back to menu.bin
and it starts the previous app, 
I have tried menu.bin from SD zip, and my own compiled version,

Thanks,
grooves

tobozo, 2018-09-27:
so did the SPIFFS copy of menu.bin work ?

`flash read err, 1000` clearly indicates a read problem, so even if this test seems unrelated and redundant, making this test work or fail will be a useful information.

>  I can select an app and can start the app

As a second useful, non redundant but also unrelated test, did you try to build a custom app rather than using the apps provided with the repo ?

Also more information is required to better understand the problem:

Model of M5Stack ?
Compiling environment ? Compilation Settings ?
OS ?
Model of the SD Card ?

grooves38, 2018-09-28:
Hi,

it is a fresh Arduino installation on windows 10 x64
and installed all required libs, you mentioned in your description,
M5Stack shows just a small label with 2018.3
SD card is a Samsung 32GB, FAT32 formatted
your assumtions is probably right, something to do with SPIFFS, 
when I start menu.bin I observe:

M5Stack initializing...OK
Scanning SPIFFS for binaries
E (760) SPIFFS: mount failed, -10025
SPIFFS Mount Failed
Listing directory: /
  DIR : /System Volume Information
  IGNORED FILE: /WeatherParams.txt
  IGNORED FILE: /Readme.txt
  IGNORED FILE: /RadioStations.txt
  FILE: /Tetris.bin
  DIR : /jpg
  DIR : /json
  DIR : /mp3
  FILE: /arduinomegachess.bin
  FILE: /Colours_Demo.bin

Anything I can do to solve that problem ?
have already erased the whole flash via esptool, and ESPFlashDownload

Regards,
grooves

tobozo, 2018-09-28:
`SPIFFS mount failed` can happen after a full erase or when the ESP32 is brand new, you can use the [ESP32 Sketch Data Upload](https://github.com/me-no-dev/arduino-esp32fs-plugin) once to fix that.

Running this tool will upload the contents of /data/ into your SPIFFS partition, just leave the /data/ folder empty for formatting.

[image]

Also make sure you allocate some space for OTA in the flash settings (the sd-updater is re-using the flashing part from this component) 

[image]

It may be worth trying with different settings since this library existed before the options appeared in the Arduino menu.

I haven't tried selecting the M5Fire profile and it looks like it doesn't have any options to setup the partition but it may be worth a try too.

I'll probably update the menu code and add some kind of partition detection + formatting once the M5 environment becomes stable.

grooves38, 2018-09-28:
Hi,
used the tool above, and SPIFFS error seems to be gone:

M5Stack initializing...OK
Scanning SPIFFS for binaries
/esp32fs.jar
No binary to transfer
Listing directory: /
  DIR : /System Volume Information
  IGNORED FILE: /WeatherParams.txt
  IGNORED FILE: /Readme.txt
  IGNORED FILE: /RadioStations.txt
  FILE: /Tetris.bin
  DIR : /jpg
  DIR : /json
  DIR : /mp3
  FILE: /arduinomegachess.bin
  FILE: /Colours_Demo.bin

but, it doesn't help for initial issue, can't go back to menu.bin
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1496
load:0x40078000,len:8596
load:0x40080400,len:6980
entry 0x400806f4
M5Stack initializing...OK
Will Load menu binary
Try to start update
percent = 0
percent = 1
.....
percent = 71
percent = 72
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1496
load:0x40078000,len:8596
load:0x40080400,len:6980
entry 0x400806f4
M5Stack initializing...OK

COLOURS test!

and I'm back in previous app

Regards,
Michael

tobozo, 2018-09-28:
the menu.bin should be compiled + copied onto the SD Card + also flashed with the new settings and the observed failure can only be valid if you saw it happen using any of the three different options for partition scheme.

BTW is your model a M5Stack classic, M5Fire or M5Go ?

tobozo, 2018-09-28:
Also if you don't mind gitter.im. there's a [M5Stack Fans lobby](https://gitter.im/M5STACK_FANS/Lobby) where we can chat and look further what's going on with your build without spamming this issue.

tobozo, 2018-09-28:
Found the probably source of the problem after only one beer : SPIFFS partition scheme mismatching (with an earlier version of the SDK ?).

Updating the ESP32 SDK apparently did the trick.

Closing this issue as solved, feel free to open a new issue for any esp-idf inquiry :+1:

closed 2018-09-28 by tobozo