tobozo/M5Stack-SD-Updater #32: Compile error of %s augument

tomorrow56, 2019-03-21:
Hello,

When I compile version 0.4.1 with AuruinoIDE, get the following error.

Arduino\libraries\M5Stack-SD-Updater\src\M5StackUpdater.cpp:212:91: error: format '%s' expects argument of type 'char*', but argument 3 has type 'const __FlashStringHelper*' [-Werror=format=]

     Serial.printf( "[M5Stack-SD-Updater] SD Updater version: %s\n", M5_SD_UPDATER_VERSION );

 Casting to .char * will eliminate the error. Is this correct?
    Serial.printf( "[M5Stack-SD-Updater] SD Updater version: %s\n", (char*)M5_SD_UPDATER_VERSION );

Thank you.

tobozo, 2019-03-21:
Hello, thank you for reporting this.

Apparently this only occurs when "all" is selected in Arduino compilation warning preferences.

[image]

I've applied your suggested fix for `M5_SD_UPDATER_VERSION` and also for `M5_LIB_VERSION`.

Thanks a lot for your contribution 🙇 👍

tomorrow56, 2019-03-23:
Thank you very much!!

closed 2019-03-23 by tomorrow56