tobozo/M5Stack-SD-Updater #44: [test] [SD-App] please add M5Stack_LovyanToyBox.

lovyan03, 2019-04-29:
Please add this project to the m5stack registry  https://github.com/lovyan03/M5Stack_LovyanToyBox

tobozo, 2019-04-29:
Thank you for your submission.
This project is now awaiting validation.

tobozo, 2019-04-29:
adding library requirements: 

    https://github.com/kosme/arduinoFFT/
    https://github.com/lovyan03/M5Stack_TreeView/
    https://github.com/lovyan03/M5Stack_OnScreenKeyboard/

tobozo, 2019-05-14:
[image]
This message is an automated reply:

The Travis CI build passed with https://github.com/lovyan03/M5Stack_LovyanToyBox.
See https://travis-ci.org/tobozo/M5Stack-SD-Updater/builds/532318768 for more info.

tobozo, 2019-08-29:
pre-hook:

    sed -i -e 's/<arduino\.h>/<Arduino\.h>/g' LovyanToyBox/src/tjpgdClass.cpp
    # keeping this hook until the original source is fixed (ping @lovyan03)

tobozo, 2019-08-29:
[image]
This message is an automated reply:

The Travis CI build passed with https://github.com/lovyan03/M5Stack_LovyanToyBox.
See https://travis-ci.org/tobozo/M5Stack-SD-Updater/builds/578144506 for more info.

tobozo, 2019-08-29:
[image]
This message is an automated reply:

The Travis CI build passed with https://github.com/lovyan03/M5Stack_LovyanToyBox.
See https://travis-ci.org/tobozo/M5Stack-SD-Updater/builds/578145341 for more info.