tobozo/M5Stack-SD-Updater #46: [SD-App] please add M5Chorder

wararyo, 2019-05-16:
Please add this project to the m5stack registry https://github.com/wararyo/M5Chorder

tobozo, 2019-05-16:
[image]
This message is an automated reply:

The Travis CI build passed with https://github.com/wararyo/M5Chorder.
See https://travis-ci.org/tobozo/M5Stack-SD-Updater/builds/533268346 for more info.

tobozo, 2019-05-19:
[image]
This message is an automated reply:

The Travis CI build passed with https://github.com/wararyo/M5Chorder.
See https://travis-ci.org/tobozo/M5Stack-SD-Updater/builds/533278526 for more info.