tobozo/M5Stack-SD-Updater #121: [SD-App] please add @G4lile0's ESP32-WiFi-Hash-Monster

tobozo, 2019-11-18:
https://github.com/G4lile0/ESP32-WiFi-Hash-Monster

tobozo, 2019-11-18:
pre-hook:

    wget https://raw.githubusercontent.com/m5stack/M5Stack/master/examples/Advanced/Display/Free_Font_Demo/Free_Fonts.h --output-document=ESP32-WiFi-Hash-Monster/Free_Fonts.h

post-hook:

    #

tobozo, 2019-11-18:
add library requirement:

    https://github.com/FastLED/FastLED/archive/3.3.2.zip

tobozo, 2019-11-18:
The app has been scheduled for rebuild

tbaumann, 2020-09-24:
I recently added platform.io support to that project. Should be merged soon. This might help you with building.

However the builds are platform dependent. How do you deal with that in general? There are multiple variants of M5Stack and odroid

tobozo, 2020-09-24:
with the help of LovyanGFX driver autodetect feature and a few additions to the ESP32-Chimera-Core it will soon be possible to have a single binary that fits all devices, so the future is very bright

the real issue with multiple variants is about the partitions schemes, mixing them is still possible but it can prevent the OTA rollback to work properly and this makes the SD-Updater very slow to load binaries