tobozo/M5Stack-SD-Updater #134: Can't turn off M5Stack Grey

chrwh, 2020-04-24:
Sorry if this is not an SD-Updater bug. But I'm having trouble turning off the M5Stack by double clicking the side button. When first powered on you can turn it off without any issues, but when the M5Stack has been auto powered off by SD-Updater , and then turned on with button B, it's no longer possible to turn it off by double clicking the side button?
Only after removing the battery it works one time again.

tobozo, 2020-04-24:
Thanks for reporting this :+1
 
I'll be testing that with a M5 Classic after the battery is charged, I just remembered the double push on side button only works when the USB isn't plugged in :facepalm: 

But I can't see why the SD-Updater library would do that.
So I assume you're probably talking about the `M5Stack-SD-Menu` sketch from the examples folder ? 
This one uses `M5.powerOff()` in a standard way, and may have some bugs, but there are also hardware bugs on some models.

[edit] I need to fully charge the battery first, could you test the same scenario with that code and confirm that the problem occurs (or does not occur) ?

```C

#include <M5Stack.h>
#include "M5StackUpdater.h"

void setup() {
  M5.begin();
  Wire.begin();
  if(digitalRead(BUTTON_A_PIN) == 0) {
    Serial.println("Will Load menu binary");
    updateFromFS(SD);
    ESP.restart();
  }
  M5.Lcd.fillScreen( TFT_GREEN );
  delay(1000);
}
void loop() {
  M5.update();
  if( M5.BtnB.wasPushed() ) {
    M5.setWakeupButton( BUTTON_B_PIN );
    M5.powerOff();
  }
  delay(100);
}
```

chrwh, 2020-04-24:
Thanks for the answer, yes I'm using the M5Stack-SD-Menu.
I had to modify the code a bit to get it to compile. When pressing button B it just reboots?

`#include <M5Stack.h>
#include "M5StackUpdater.h"

void setup() {

  M5.begin();
  M5.Power.begin(); //Added
  Wire.begin();

  if(digitalRead(BUTTON_A_PIN) == 0) {

    Serial.println("Will Load menu binary");

    updateFromFS(SD);

    ESP.restart();

  }

  M5.Lcd.fillScreen( TFT_GREEN );

  delay(1000);

}

void loop() {

  M5.update();

  if( M5.BtnB.wasPressed() ) { // Changed from wasPushed

    M5.Power.setWakeupButton( BUTTON_B_PIN );//Added Power

    M5.Power.powerOFF(); //Added Power and changed poweroff to powerOFF

  }

  delay(100);

}
`

chrwh, 2020-04-24:
Thanks for the edit, I could not get the code part to look right, I tried to click the <> but did not have much luck with that

tobozo, 2020-04-24:
of course if the wakeup button is the same as the sleep button, it wakes up immediately

sorry that was stupid of me, using `M5.BtnC.wasPressed()` instead should fix that

chrwh, 2020-04-24:
Ha ha - sure, should have caught that one ;-)
But the issue is still the same with or with out the usb connected. Should I load the menu or thry from the greenscreen?

tobozo, 2020-04-24:
First unplug the USB, because the power module has a different behaviour when the M5 is charging.

When you see the greenscreen, hit button C to make it go to sleep.

Then you can test if the double push on the side button wakes it up.

If you observe the same problem, then it's likely your M5 Power module is the cause.

tobozo, 2020-04-24:
So after a full charge here's the behaviour on my M5Classic grey:

1) unplugged usb (it turned the M5 on an started the SD-Menu)
2) waited for auto sleep
3) pushed button A, B, C => no effect
4) double push reset => M5 wakes up
5) double push reset (after UI load) => M5 goes to sleep

chrwh, 2020-04-24:
Well I can't make it turn off with button C I have tried the same code on 2 different Core grey. But when double push the side button they turn off.
Tried what you described above, 
1. OK
2. OK
3. Button B turned it on
4. Well it's allready on
5. Can't turn it off again

tobozo, 2020-04-24:
so you do have two different core grey, what about swapping power modules and test again ?

If those two m5 aren't from the exact same generation, the power modules may have a different pinout, and also different bugs :-)

chrwh, 2020-04-24:
Tried that already, one is the simple battery bottom and the other one is faces bottom, same thing, the m5 does not have same date code 1 is 2018.3 and the other one is 2018.2A. They can both talk i2c to the IP5306 chip. 
I'm a bit lost here

tobozo, 2020-04-24:
looks like the IP5306 hasn't always been wired to I2C

https://github.com/m5stack/M5Stack/issues/156

since I don't have the right hardware combination to reproduce that particular issue, I can only suggest ideas, if you're okay with testing, here's a track you could follow:

```C
if(!M5.Power.canControl()) {
  M5.Lcd.printf("IP5306 is not i2c version\n");
  esp_sleep_enable_timer_wakeup(30e6);
  esp_deep_sleep_start();
} else {
  M5.Lcd.printf("IP5306 is i2c version\n");
  M5.powerOff();
}
```

chrwh, 2020-04-24:
Sure I'll test. I'm still working with the M5Stack-SD-Menu where should I put the code?

tobozo, 2020-04-24:
if you're working with the menu code you may want to put a lower value to `MS_BEFORE_SLEEP`

Here's where the autosleep is triggered: https://github.com/tobozo/M5Stack-SD-Updater/blob/master/examples/M5Stack-SD-Menu/menu.h#L850

chrwh, 2020-04-24:
Did the change, still the same

chrwh, 2020-04-25:
I have written M5 support it seems like it could be and PMU issue, since it's working when the battery power has been removed, then it works again for a while

tobozo, 2020-06-08:
hey @chrwh 

it could be interesting to check how using any of those affects the current behaviour:

    WiFi.setSleep(false);

or 

    WiFi.setSleep(true);

tobozo, 2021-09-10:
Closing this issue as it's getting old, also M5Classic is now facing EOL and it makes this rare bug out of scope.

Feel free to continue this conversation though, I still have a couple of those and can eventually compare behaviours.

closed 2021-09-10 by tobozo