tobozo/M5Stack-SD-Updater #138: Cant get to work on a Core2

Joniss77, 2020-09-15:
It compile and loads to the Core2 but the display stays black? Any idea´s?

best regards
Joniss77

tobozo, 2020-09-15:
hey thanks for your feedback

A few things to check in order to enable M5Core2 support.

1) you're using the "M5Stack-Core2" choice from the Arduino Boards menu
2) you're using the new `checkSDUpdater();` function to check for 'button' press
3) you're using either ESP32-Chimera-Core, or Lovyan03'2 [M5Core2](https://github.com/lovyan03/M5Core2) fork (until [this pull request[(https://github.com/m5stack/M5Core2/pull/9) is complete)
4) you touch the display 'button' as soon as the small square appears at boot (lasts 500ms)

by flashing your M5Core2 with Core Debug Level "Warn" there will eventually be some extra information in the Serial output window.

Joniss77, 2020-09-22:
Thanks for your reply, I got everting working now. 👍

tobozo, 2020-09-22:
that's good news :-)

closing this issue as solved, feel free to reopen if necessary

closed 2020-09-22 by tobozo