tobozo/M5Stack-SD-Updater #142: [Documentation] WiFi usage is not documented

tbaumann, 2020-09-24:
Apparently M5Stack-SD-Updater can download stuff via WiFi.

However it's not described how WiFi is set up. It just times out.

tbaumann, 2020-09-24:
I suspect it just works when PSRAM Preferences `WIFI_SSID` and `WIFI_PASSWD` are set correctly.

I managed to set them up via `SetWiFi_Mic`.

There is also https://github.com/m5stack/M5Stack/blob/master/examples/Advanced/WiFiSetting/WiFiSetting.ino
That should be included in the SD Card content.

And the Updater should detect it if whose Preferences are missing and explain (on screen) how to set them.

tobozo, 2020-09-24:
hey @tbaumann thanks for your feedback

There's no UI for WiFi setup, the launcher just assumes your ESP32 had a previous successful connection to a WiFi network and will re-use these credentials transparently.

Flashing a basic WiFi sketch and connecting once successfully, then reflashing the launcher example should do it.

tbaumann, 2020-09-24:
Graceful fail and some explanation how to fix it would help a lot in case there is no WiFi config. ;)

Thanks

tobozo, 2021-09-10:
Closing this as wontfix, wifi manager can still be used an an external application and is better kept outside the SD-Menu.

Feel free to add some followup e.g. if you have a specific WiFi-Manager to recommend / test.

closed 2021-09-10 by tobozo