tobozo/M5Stack-SD-Updater #143: [SD-App] Please add simple WiFi Setup app

tbaumann, 2020-09-24:
The out of the box experience is pretty poor if there is no WiFi Setup in PSRAM.

Please add this simple WiFi Setup script to the application list to allow WiFi setup for new users

https://github.com/m5stack/M5Stack/blob/master/examples/Advanced/WiFiSetting/WiFiSetting.ino

tobozo, 2020-09-24:
This is a great idea :+1:  

I'm testing the M5Stack WiFiSettings example now and found out that this sketch is not helpful when the SSID of the AP is hidden (WPA/PSK).

So I've just pushed the changes as they are and will wait for your feedback on this before going further.

Next steps are to provide a proper UI and, extend the webserver to allow the user to manually enter a hidden SSID, show the mac address for easier insertion in allow-lists, and eventually give a better css to this page :-)

tbaumann, 2020-09-25:
Nice, direct integration in the app is even nicer.

tobozo, 2021-09-10:
bump

The arduino-esp32 core 2.0 brought some improvements but also made the resulting binary size unfit for default partition and uses so much ram the WiFi becomes unusable.

Since core 2.0, the only devices able to run the Downloader and WiFi operations must have PSRam and a bigger partition, this eliminates basic ESP32-Wroom support (e.g M5Stack gray), and only leaves M5Fire and M5Core2 family devices as supported.

The only one-solution-for-all I can think about is to have those WiFi-Manager/Downloader features moved to separate binaries, and chain them as pass-through apps instead of the usual loading scenario (e.g. replace menu.bin by the *next* app).

I'll close this issue after I manage to get the Downloader isolated to a separate app.

be well