tobozo/M5Stack-SD-Updater #146: M5Stack-SD-Updater does not work on a M5Stack-Faces (SD-Card problem)

blaumg, 2020-10-01:
I use the M5stack faces. When starting the m5stack-sd-menu, the message "Insert SD .. Checking SD Card ..." appears on the display. There is a 16 GB SD card in the device. When I detach the M5Stack from the Faces-Base, the software works fine. What can you do?

blaumg, 2020-10-01:
- M5Stack model gray: „M5Stack Faces Kit“ with 3 keyboards included (keyboard panel, calculator panel and gamepad panel). 

- The problem occurs with each of the 3 keyboards and also without them. Simply plugging the M5Core onto the bottomboard causes the problem.

- I use the unchanged program code from the Github page.

#include "menu.h"
void setup() {
  #if defined(_CHIMERA_CORE_)
    M5.begin(true, false, true, false, false); // bool LCDEnable, bool SDEnable, bool SerialEnable, bool I2CEnable, bool ScreenShotEnable
  #else
    M5.begin(); // bool LCDEnable, bool SDEnable, bool SerialEnable, bool I2CEnable, bool ScreenShotEnable
  #endif
  #if defined HAS_TOUCH
    // suggest rollback
    checkSDUpdater( M5_FS, "" );
  #endif
  //WiFi.onEvent(WiFiEvent); // helps debugging WiFi problems with the Serial console
  UISetup(); // UI init and check if a SD exists
  doFSChecks(); // replicate on SD and app1 partition, scan data folder, load registry
  doFSInventory(); // enumerate apps and render menu
}
void loop() {
  HIDMenuObserve();
  sleepTimer();
}

tobozo, 2020-10-01:
Thanks to you I spotted a battery problem on my M5Faces, it's all inflated :-(

[image]

Now time for debugging. 

Do you know for sure this M5Faces kit works with this particular M5Stack ? 
If so can you share the sketch so that we can align and conduct further tests.

If you can't get M5Faces to work alone, the best would be to try with another M5Stack, the pinout changed over time, and there are some incompatibilities.

You should expect to see M5Faces in an I2C scan result (0x08 is the ID for the keyboard) :

```
      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
              -- -- -- -- -- 08 -- -- -- -- -- -- --
 01: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
 02: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
 03: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
 04: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
 05: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
 06: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
 07: -- -- -- -- -- -- -- --

```

blaumg, 2020-10-01:
Thanks for the quick reply. 
I bought the M5Stack Faces Kit as a complete package. So that should fit together.
I downloaded the sketch from https://github.com/tobozo/M5Stack-SD-Updater.git. Unfortunately I only have one M5 stack, so I cannot check.

The following I2C Addresses were found:

**M5Stack without Faces base**
I2C device found at address 0x10  !
I2C device found at address 0x68  !
I2C device found at address 0x75  !
>>>> Found total 3 devices

 **Faces without any panel attached**
I2C device found at address 0x10  !
I2C device found at address 0x68  !
I2C device found at address 0x75  !
>>>> Found total 3 devices

**Faces with keyboard panel**
I2C device found at address 0x08  !
I2C device found at address 0x10  !
I2C device found at address 0x68  !
I2C device found at address 0x75  !
>>>> Found total 4 devices

**Faces with gamepad panel**

I2C device found at address 0x08  !
I2C device found at address 0x10  !
I2C device found at address 0x68  !
I2C device found at address 0x75  !
>>>> Found total 4 devices

**Faces with calculator panel**
I2C device found at address 0x08  !
I2C device found at address 0x10  !
I2C device found at address 0x68  !
I2C device found at address 0x75  !
>>>> Found total 4 devices

tobozo, 2020-10-02:
0x75 is the IP5306 power module (the M5Fire has one too), so it's a different M5Stack model from the one I have.

Did you manage to get the M5Faces example sketch to work so you can confirm the module isn't faulty ?

blaumg, 2020-10-10:
In the meantime I got a complete exchange of the M5Stack Faces. But the problem is still there. I can assume the last part was okay too. The example sketches all work.

tobozo, 2020-10-11:
I found my other M5Stack, both are gray but different generation.

The one on the right hand side does not show any SD Card problem so I'll try with the older one.

[image]

Now we need to agree on a common sketch code.

Here's the sketch I've been testing successfully (M5Faces responds normally and the SD-Updater successfully loads the menu.bin when the BtnA is pushed at boot time): 

```C++
#include <ESP32-Chimera-Core.h>
#include "drivers/M5Stack/M5Faces.h"
#include <M5StackUpdater.h>

M5Faces Faces;

#define GAMEBOY_KEY_NONE        0x00
#define GAMEBOY_KEY_RELEASED    0xFF
#define GAMEBOY_KEY_START       0x7F
#define GAMEBOY_KEY_SELECT      0xBF
#define GAMEBOY_KEY_A           0xEF
#define GAMEBOY_KEY_B           0xDF
#define GAMEBOY_KEY_UP          0xFE
#define GAMEBOY_KEY_DOWN        0xFD
#define GAMEBOY_KEY_LEFT        0xFB
#define GAMEBOY_KEY_RIGHT       0xF7

uint8_t lastReleasedKey = 0x00;

void onKeyPushed(const char* msg) {
  Serial.println(msg);
  lastReleasedKey = GAMEBOY_KEY_NONE;
}

void setup() {
  M5.begin();
  checkSDUpdater();
  Wire.begin(SDA, SCL);
  if( !Faces.canControlFaces() ) {
    Serial.println("This driver can't control M5Faces, aborting");
    while(1);
  }
}

void loop() {
  uint8_t keypadState = Faces.getch();
  // look for "released" signal
  if( keypadState == GAMEBOY_KEY_RELEASED ) {
    keypadState = lastReleasedKey;
  } else {
    lastReleasedKey = keypadState;
    keypadState = GAMEBOY_KEY_NONE;
  }

  switch( keypadState ) {
    case GAMEBOY_KEY_UP :
      onKeyPushed("Key Up");
    break;
    case GAMEBOY_KEY_RIGHT :
      onKeyPushed("Key Right");
    break;
    case GAMEBOY_KEY_DOWN :
      onKeyPushed("Key Down");
    break;
    case GAMEBOY_KEY_LEFT :
      onKeyPushed("Key Left");
    break;
    case GAMEBOY_KEY_START :
      onKeyPushed("Key Start");
    break;
    case GAMEBOY_KEY_SELECT :
      onKeyPushed("Key Select");
    break;
    case GAMEBOY_KEY_A :
      onKeyPushed("Key A");
    break;
    case GAMEBOY_KEY_B :
      onKeyPushed("Key B");
    break;
    default:
      delay(100);
  }
}

```

It was compiled with the following settings:

- Board: M5Stack-Core-ESP32
- Core debug level: Warn

and produced the following output in the serial console:

[image]

blaumg, 2020-10-11:
I got the following compile errors:

Arduino: 1.8.13 (Windows 10), Board: "M5Stack-Core-ESP32, QIO, 80MHz, Default, 921600, Warn"

C:\Users\Blaum\Documents\Arduino\libraries\ESP32-Chimera-Core\src\helpers\ScreenShotService\PNG\FatPNGEncoder.cpp: In member function 'bool PNG_Encoder::encodeToFile(const char*, int, int)':

C:\Users\Blaum\Documents\Arduino\libraries\ESP32-Chimera-Core\src\helpers\ScreenShotService\PNG\FatPNGEncoder.cpp:62:115: error: 'tdefl_write_image_to_png_file_in_memory_ex' was not declared in this scope

   void *PNGDataPtr = tdefl_write_image_to_png_file_in_memory_ex(rgbBuffer, imageW, imageH, 3, &png_data_size, 6, 0);
                                                                                                                 ^

Mehrere Bibliotheken wurden für "SD.h" gefunden

 Benutzt: C:\Users\Blaum\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD

 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\SD

 Nicht benutzt: C:\Users\Blaum\Documents\Arduino\libraries\SD

exit status 1

Fehler beim Kompilieren für das Board M5Stack-Core-ESP32.

tobozo, 2020-10-11:
you probably need to update LovyanGFX to the latest version

blaumg, 2020-10-11:
you're right. I've updated the LovyanGFX and teh result is here:

21:06:48.277 -> ets Jun  8 2016 00:22:57
21:06:48.277 -> 
21:06:48.277 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
21:06:48.277 -> configsip: 0, SPIWP:0xee
21:06:48.277 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:06:48.277 -> mode:DIO, clock div:1
21:06:48.277 -> load:0x3fff0018,len:4
21:06:48.277 -> load:0x3fff001c,len:1216
21:06:48.277 -> ho 0 tail 12 room 4
21:06:48.277 -> load:0x40078000,len:9720
21:06:48.277 -> ho 0 tail 12 room 4
21:06:48.277 -> load:0x40080400,len:6352
21:06:48.277 -> entry 0x400806b8
21:06:48.546 -> ESP32-Chimera-Core initializing...[W][ESP32-Chimera-Core.cpp:224] sd_begin(): Enabling SD from TFCARD_CS_PIN #4 at 20000000 Hz
21:06:48.546 -> [E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(1)
21:06:48.546 -> [W][LGFX_Config_AutoDetectESP32.hpp:230] init_impl(): [Autodetect] load from NVS : board:2
21:06:48.581 -> [W][LGFX_Config_AutoDetectESP32.hpp:408] init_impl(): [Autodetect] panel id:000000e3
21:06:48.581 -> [W][LGFX_Config_AutoDetectESP32.hpp:448] init_impl(): [Autodetect] M5Stack
21:06:48.954 -> [W][ESP32-Chimera-Core.cpp:224] sd_begin(): Enabling SD from TFCARD_CS_PIN #4 at 20000000 Hz
21:06:48.988 -> [E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(1)
21:06:48.988 -> OK
21:06:48.988 -> [E][M5StackUpdaterUI.h:318] checkSDUpdaterUI(): Booting with reset reason: 1
21:08:22.111 -> Key A
21:08:23.037 -> Key B
21:08:25.130 -> Key Up
21:08:27.254 -> Key Start
21:08:27.938 -> Key Select
21:08:31.655 -> Key Up
21:08:31.962 -> Key Up
21:08:33.887 -> Key Down

blaumg, 2020-10-11:
[image]

blaumg, 2020-10-11:
output from console without Faces Bottom:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
ESP32-Chimera-Core initializing...[W][ESP32-Chimera-Core.cpp:224] sd_begin(): Enabling SD from TFCARD_CS_PIN #4 at 20000000 Hz
[W][LGFX_Config_AutoDetectESP32.hpp:230] init_impl(): [Autodetect] load from NVS : board:2
[W][LGFX_Config_AutoDetectESP32.hpp:408] init_impl(): [Autodetect] panel id:000000e3
[W][LGFX_Config_AutoDetectESP32.hpp:448] init_impl(): [Autodetect] M5Stack
OK
[E][M5StackUpdaterUI.h:318] checkSDUpdaterUI(): Booting with reset reason: 1
This driver can't control M5Faces, aborting

tobozo, 2020-10-11:
I even tried removing the M5Fire bottom but none of my M5stack version seems to be able to replicate this situation :-(

Some random things I can think about: 

- Showing me an example sketch where SD and M5Faces work together on your build 
- Narrowing down the faulty part, do the sd_errors still occur when:
  - No gameboy or other module except the M5Stack is present on the M5Faces ?
  - The M5Faces battery is unplugged

Spot mechanical failures:
- See if the soldering on both M-Bus connectors (M5Faces and M5Stack) isn't damaged, like the hex cap screws on your M5core :-)
- Try another SD Card
- Ask on the M5Stack forum if there are known problems between "2018.3" Core and M5Faces

tobozo, 2021-09-10:
Closing this as out of scope, tried different combinations of M5Faces/M5*Cores and found out there are so many versions of this (and so many aren't compatible with each other) that it's up to the maker to find a working combination.

Feel free to reopen if you get a working case of M5Faces/M5Stack and still face the same issue.

closed 2021-09-10 by tobozo