tobozo/M5Stack-SD-Updater #148: Can't compile M5Stack-SD-Updater example -

Gestas, 2020-10-07:
I'm trying to compile the M5Stack-SD-Updater example, I'm getting a set of errors - 

```
In file included from /home/user/Arduino/libraries/M5Stack-SD-Updater-master/src/M5StackUpdater.h:127:0,
                 from /tmp/arduino_build_560249/sketch/menu.h:74,
                 from /home/user/Arduino/libraries/M5Stack-SD-Updater-master/examples/M5Stack-SD-Menu/M5Stack-SD-Menu.ino:80:
/home/user/Arduino/libraries/M5Stack-SD-Updater-master/src/M5StackUpdaterUI.h: In function 'void freezeTextStyle()':
/home/user/Arduino/libraries/M5Stack-SD-Updater-master/src/M5StackUpdaterUI.h:110:36: error: 'class M5Display' has no member named 'textsize'
     SDUTextStyle.textsize    = tft.textsize;
                                    ^
/home/user/Arduino/libraries/M5Stack-SD-Updater-master/src/M5StackUpdaterUI.h:111:36: error: 'class M5Display' has no member named 'textdatum'
     SDUTextStyle.textdatum   = tft.textdatum;
                                    ^
/home/user/Arduino/libraries/M5Stack-SD-Updater-master/src/M5StackUpdaterUI.h:112:36: error: 'class M5Display' has no member named 'textcolor'
     SDUTextStyle.textcolor   = tft.textcolor;
                                    ^
/home/user/Arduino/libraries/M5Stack-SD-Updater-master/src/M5StackUpdaterUI.h:113:36: error: 'class M5Display' has no member named 'textbgcolor'
     SDUTextStyle.textbgcolor = tft.textbgcolor;
                                    ^
```
I'm using the newest version of this project and these libraries -
```
ArduinoJson
LovyanGFX-master
ESP32_Chimera_Core-1.1.0
M5Core2-master
FastLED-master
M5Stack-SD-Updater-master
```

tobozo, 2020-10-07:
Hello and thanks for your detailed feedback.

I'm currently updating ESP32-Chimera-Core to cope with some recent changes in the LGFX driver, the [master branch](https://github.com/tobozo/ESP32-Chimera-Core) should compile without errors.

Please let me know if that fixed it for you and I'll create a new release.

Gestas, 2020-10-08:
@tobozo I can confirm that using the master branch (https://github.com/tobozo/ESP32-Chimera-Core/commit/971509d2e075289198cdcad56b31ed7c1b0855a6) of [ESP32-Chimera-Core](https://github.com/tobozo/ESP32-Chimera-Core) I can build the example. Thanks!

tobozo, 2020-10-08:
cool, thanks for that

I've produced a new release of the chimera-core and it'll land soon in the Arduino Library Manager

closing this issue as solved

closed 2020-10-08 by tobozo