tobozo/M5Stack-SD-Updater #156: error compiling

fermintm, 2021-09-12:
Hi! Arduino IDE 1.8.15 and esp32 v1.0.6 or 2.0.0 version same error.

M5StackUpdater.h:152:88: error: missing binary operator before token "("
       #if defined ESP_ARDUINO_VERSION && ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(2,0,0);

tobozo, 2021-09-12:
hey @fermintm thanks for reporting that :+1: 

I've pushed an update to cover that compilation error and will produce a release soon.

Meanwhile it can be fixed by manually replacing the offending macro with this:

    #if defined ESP_IDF_VERSION_MAJOR && ESP_IDF_VERSION_MAJOR >= 4

... or by updating ESP32 package to version 2.0, where this new macro was introduced.

tobozo, 2021-09-20:
closing this as fixed in latest release ([v1.1.2](https://github.com/tobozo/M5Stack-SD-Updater/tree/v1.1.2))

feel free to reopen if you still get a compilation error

thanks again for reporting :+1:

closed 2021-09-20 by tobozo