tobozo/M5Stack-SD-Updater #179: __has_include breaks under some conditions

GOB52, 2023-05-02:
* Using master 1.2.4 
* iMac macOS 10.15.7 (Catalina)
* VsCode 1.77.3
* PlatformIO Core, version 6.1.6
---

M5StackUpdater.hpp
```cpp
#if ! SDU_HAS_FS
  #define SDU_HAS_SD
  #if defined USE_SDFATFS
    #pragma message "SDUpdater will use SdFat"
    #undef __has_include <======================= !!!
    #include <misc/sdfat32fs_wrapper.h>
    #define __has_include <======================= !!!
    #define SDU_SD_BEGIN SDU_SDFatBegin
```

If the above part has been  compiled, 
A compile error occurs at subsequent __has_include.

```cpp
#include <M5StackUpdater.hpp>
#if __has_include(<foo.hpp>) <==  error: operator '<' has no left operand
#endif
```

tobozo, 2023-05-02:
there may be a fix in 1.2.5, it's addressing a similar issue, however the fix is only for an error with ESP32-targz when gzip is enabled and SdFat is used.

the workaround is to evaluate the `__has_include` earlier in the flow:

```c
#if __has_include(<foo.hpp>)
  #define HAS_FOO
#endif
#include <M5StackUpdater.h>

#if defined HAS_FOO
  // foo specific code
#endif 
```

I'm aware it isn't a satisfying solution and still researching, any suggestion is welcome.

GOB52, 2023-05-02:
This workaround is not a panacea, but it is one solution.

Hopefully 1.2.5 will be a smarter solution.

tobozo, 2023-05-02:
SdFsFileImpl is the closest thing I could find, and it came with the infamous `#undef __has_include`, so maybe it's just optional?

Arduino IDE doesn't seem to complain when I do this, but I suspect platformio will not:

```cpp
    //#undef __has_include
    #include <FS.h>
    #include <FSImpl.h>
    #include <SdFat.h>
    //#define __has_include
```

the smarter solution is probably to create some kind of lgfx::DataWrapper for SdFs, but it is beyond my skills.

GOB52, 2023-05-02:
I know it's risky, but how about this approach? 

```cpp
#undef __has_include
#include <misc/sdfat32fs_wrapper.h>
#define __has_include(STR)  __has_include__(STR)
```
__has_include is defined by  \_\_has_include\_\_

tobozo, 2023-05-02:
looks like platformio doesn't care when the infamous `#undef __has_include` is removed :thinking:  so I may remove it completely?

https://github.com/tobozo/M5Stack-SD-Updater/actions/runs/4863851743

PR #181 (2023-05-03): 1.2.5
- fixes #180
- fixes #179 
- fix for `SDUCfg.setBinFileName()` crashing
closed 2023-05-03 by tobozo