tobozo/M5Stack-SD-Updater #183: AXP issue

Koekebeest, 2023-06-13:
Hello,

I'm trying to flash the SD updater on my m5core2 but I keep running into various issues. The errors has no member named 'Axp' suggest that the library you are using (ChimeraCore::ECCKernel) does not have a member named 'Axp and a lot of conflicting libraries for various .h files. What am I doing wrong here? Hope someone can guide me in the right direction. Would love to be able to have multiple apps on the device.

Koekebeest, 2023-06-13:
Hi there. First of all I'm amazed at the work you do. I'm a total noob so let me try and answer you with my limited knowledge. 
Platform name: Arduino Cloud on M5core2
OS: Arduino
Core version: ESP32 D0WDQ6-V3 (m5core2)
Library versions:

 // M5GFX - Version: 0.1.7
#include <M5GFX.h>

// M5Unified - Version: 0.1.7
#include <M5Unified.h>

// M5Core2 - Version: 0.1.5
#include <M5Core2.h>

// LovyanGFX - Version: 1.1.7
#include <LovyanGFX.hpp>

// ESP32-Chimera-Core - Version: 1.4.9
#include <ESP32-Chimera-Core.h>

MRE:

enum HIDSignal
{
  HID_INERT = 0,
  HID_UP = 1,
  HID_DOWN = 2,
  HID_SELECT = 3,
  HID_PAGE_DOWN = 4,
  HID_PAGE_UP = 5,
  HID_SCREENSHOT = 6
};

static bool M5FacesEnabled = false;

#ifdef CAN_I_HAZ_M5FACES
#include <M5Faces.h>
M5Faces Faces;
#endif

void HIDInit()
{
#ifdef CAN_I_HAZ_M5FACES
  M5FacesEnabled = Faces.canControlFaces();
#endif
  // Initialize any other required components or configurations
}

HIDSignal getControls()
{
  if (M5FacesEnabled)
  {
    // Check M5Faces inputs and return corresponding HIDSignal
    // Replace the logic below with the actual implementation
    if (Faces.getch() == GAMEBOY_KEY_UP)
      return HID_UP;
    else if (Faces.getch() == GAMEBOY_KEY_DOWN)
      return HID_DOWN;
    else if (Faces.getch() == GAMEBOY_KEY_SELECT)
      return HID_SELECT;
    else if (Faces.getch() == GAMEBOY_KEY_RIGHT)
      return HID_PAGE_DOWN;
    else if (Faces.getch() == GAMEBOY_KEY_LEFT)
      return HID_PAGE_UP;
  }
  
  // If M5Faces is not enabled or no input detected, return HID_INERT
  return HID_INERT;
}

void setup()
{
  // Initialize necessary components or configurations
  HIDInit();
}

void loop()
{
  // Read controls and handle the corresponding HIDSignal
  HIDSignal control = getControls();

  // Handle the control according to the received HIDSignal
  // Replace the logic below with the actual implementation
  switch (control)
  {
    case HID_UP:
      // Perform action for HID_UP
      break;
    case HID_DOWN:
      // Perform action for HID_DOWN
      break;
    case HID_SELECT:
      // Perform action for HID_SELECT
      break;
    case HID_PAGE_DOWN:
      // Perform action for HID_PAGE_DOWN
      break;
    case HID_PAGE_UP:
      // Perform action for HID_PAGE_UP
      break;
    case HID_SCREENSHOT:
      // Perform action for HID_SCREENSHOT
      break;
    case HID_INERT:
      // No control or inactive state
      break;
  }
}

To give some additional information on the error. This is a little part of the error that I'm getting using cloud.arduino.cc but it has numerous more. I must be doing something wrong... >.<

/usr/local/bin/arduino-cli compile --fqbn esp32:esp32:m5stack-core2:CPUFreq=240,DebugLevel=none,EraseFlash=none,PSRAM=enabled,PartitionScheme=default,UploadSpeed=921600 --libraries /home/builder/opt/libraries/latest --build-cache-path /tmp --output-dir /tmp/1417966823/build --build-path /tmp/arduino-build-79F1D1B0A10A9EB9AC699885BC653334 --library /mnt/create-efs/webide/cb/53/cb5368cf89341ebf0e133f7ff04800f3:gabrielbeaux/libraries_v2/M5Stack-SD-Updater --library /home/builder/opt/libraries/esp32_chimera_core_1_4_9 --library /home/builder/opt/libraries/lovyangfx_1_1_7 --library /home/builder/opt/libraries/m5core2_0_1_5 --library /home/builder/opt/libraries/m5unified_0_1_7 --library /home/builder/opt/libraries/m5gfx_0_1_7 /tmp/1417966823/M5Stack-SD-Menu

In file included from /tmp/1417966823/M5Stack-SD-Menu/main.cpp:1:

/tmp/1417966823/M5Stack-SD-Menu/menu.h:38:19: note: #pragma message: M5STACK Core2 detected

#pragma message "M5STACK Core2 detected"

^~~~~~~~~~~~~~~~~~~~~~~~

In file included from /tmp/1417966823/M5Stack-SD-Menu/menu.h:141,

from /tmp/1417966823/M5Stack-SD-Menu/main.cpp:1:

/tmp/1417966823/M5Stack-SD-Menu/controls.h: In function 'void vibrateTask(void*)':

/tmp/1417966823/M5Stack-SD-Menu/controls.h:42:10: error: 'using M5Stack = class ChimeraCore::ECCKernel' {aka 'class ChimeraCore::ECCKernel'} has no member named 'Axp'

M5.Axp.SetLDOEnable( 3,1 );

^~~

/tmp/1417966823/M5Stack-SD-Menu/controls.h:44:10: error: 'using M5Stack = class ChimeraCore::ECCKernel' {aka 'class ChimeraCore::ECCKernel'} has no member named 'Axp'

M5.Axp.SetLDOEnable( 3,0 );

^~~

In file included from /home/builder/opt/libraries/m5core2_0_1_5/src/M5Core2.h:76,

from /tmp/1417966823/M5Stack-SD-Menu/M5Stack-SD-Menu.ino:8:

/home/builder/opt/libraries/m5core2_0_1_5/src/M5Display.h:90:16: error: reference to 'jpeg_div_t' is ambiguous

jpeg_div_t scale = JPEG_DIV_NONE);

tobozo, 2023-06-13:
Since `<ESP32-Chimera-Core.h>` is a drop-in replacement for `<M5Core2.h>` library, **those libraries shouldn't be used together**, however in your logs I can read this:

```cpp
#include <M5Core2.h>
```

If you didn't explicitely include M5Core2 library, maybe remove it from your libraries folder to see if it improves?

Koekebeest, 2023-06-13:
> Since `<ESP32-Chimera-Core.h>` is a drop-in replacement for `<M5Core2.h>` library, **those libraries shouldn't be used together**, however in your logs I can read this:
> 
> ```c++
> #include <M5Core2.h>
> ```
> 
> If you didn't explicitely include M5Core2 library, maybe remove it from your libraries folder to see if it improves?

I've removed the inclusion but still the same. I can flash the m5core2 -sdloader-snippet but it only loads the "loading menu" but does nothing else. I feel I might just be doing something wrong here. Do I need to select a different board perhaps?

tobozo, 2023-06-13:
yup the snippet is just supposed to show the lobby, nothing more

I'm not sure why you still get the errors with the SD-Menu, but there's a [precompiled binary](https://github.com/tobozo/M5Stack-SD-Updater/releases/download/v1.2.7/M5Core2-Launcher-2.0.9.bin) built by github CI if you want to try.

From a beginner's point of view it may be easier to learn how to manually flash a binary than troubleshooting an issue that is probably related to the learning path.

Koekebeest, 2023-06-14:
Will check that out today. Thanks for your help so far :)

Koekebeest, 2023-06-16:
Can I ask which versions you use of the provided libs on your m5core2?

tobozo, 2023-06-16:
ChatGPT only highlights the obvious, I wouldn't trust it for proper troubleshooting yet anyway, it's still dreaming a big part of the answer and is more creative than intuitive.

your feedback made me realize there are a few anomalies in ESP32-Chimera-Core, you should see some improvements when selecting ESP32-Chimera-Core 1.4.6 instead of the latest.

if you get new errors incriminating M5StackUpdater, apply the same logic: decrement to the previous version of the library until it compiles.

Sorry about leaving my libraries in such a pitiful state, I just got overwhelmed by the release of M5CoreS3 and negligated the unit tests in the lates releases, I'll fix that soon :bow:

Koekebeest, 2023-06-16:
> ChatGPT only highlights the obvious, I wouldn't trust it for proper troubleshooting yet anyway, it's still dreaming a big part of the answer and is more creative than intuitive.
>
> your feedback made [...]

Yes you are right on that. It just copy pastes things from internet. Also incorrect things. Thanks for answering and providing helpful information! Really appreciate it. I will try these things out and hopefully I'll get it to work :) No worries mate. I'm just a noob trying to learn :)

Koekebeest, 2023-06-18:
SOrry to bother you again >.< for my clarification with "if you get new errors incriminating M5StackUpdater, apply the same logic: decrement to the previous version of the library until it compiles." you meant the M5stack-sd-updater lib? Because I think I tried like 5 older libs from each but not getting anywhere. Collect2 errors. Although the older version of chimera seemed to help me get out of the previous errors these don't seem to go away also with going to previous versions of chimera and/or m5stack-sd-updater. Do I also need to use older versions of lovyan perhaps?

tobozo, 2023-06-18:
>  you meant the M5stack-sd-updater lib

yes, if can't checkout the latest 1.5.0 version manually, select an earlier version from the library manager

LovyanGFX isn't the issue here, the issue is in the shallow description of the problem and the randomness of your attempts at fixing it. 

I'm not saying it's an invalid approach, but it's like brute force and it will statistically end up more often in failure than success. The good thing is that it builds a lot of experience, the lesser good thing is that it takes a lot of time to figure out stuff that can't be guessed and/or explained by peers.

I have managed to compile WiFi Hash Monster in Arduino IDE 1.8.19 using the following stack:

[image]

It's using espresssif core 2.0.5, both ESP32-Chimera-Core and M5Stack-SD-Updater libraries are from the latest branch on their respective respositories using git. If you want to reproduce the exact environment you'll need to use git too, or switch to vscode/platformio.

The original plaformio.ini file from the WiFiHashMonser repo has valid (although a bit old) combination of espressif core and library versions, it's using espressif core 2.0.0:

[image]

here's another one I've tested recently, it's using espressif core 2.0.5:

[image]

Koekebeest, 2023-07-20:
Cool thanks I will certainly try these out soon. Again appreciate your time and help. Cheers