tobozo/M5Stack-SD-Updater #185: CoreS3

jaylfc, 2023-10-03:
Hi,

I've flashed the M5Stack-SD-Menu to my CoreS3 but when the bin file loads it just reboots back into the menu, any ideas?

Thank you for your work!

tobozo, 2023-10-04:
hi, 

do you mean one of your loadable apps bounces back to SD-Menu when you attempt to load it?

it sounds like the action button is triggered before checkSDUpdater() is called, can you share the setup() code from that app?

I just did a test with a basic example and it behaves normally, so I'll need more info to reproduce this issue e.g. library version, platform, core options (especially Partition Scheme).

otherwise if you feel adventurous, the next release candidate is still being tested on the [1.2.8 branch](https://github.com/tobozo/M5Stack-SD-Updater/tree/1.2.8)

jaylfc, 2023-10-04:
Hi and thanks for getting back to me! I'll give that a go, to be honest it's actually the default apps from the SD Content. Maybe the apps themselves aren't compatible? I'll try with my own app later and see how it goes. Thanks again!

tobozo, 2023-10-04:
oh right, the [zip file in the readme](https://github.com/tobozo/M5Stack-SD-Updater/releases/download/v0.4.1/SD-Apps-Folder.zip) has a few precompiled applications but they're only supported M5GO/M5Fire devices and use an outdated version of espressif arduino-esp32 core where OTA partitions have a different offset.

I will remove this link from the readme, sorry for the confusion

jaylfc, 2023-10-04:
Ok good stuff, so it's just compatibility with the bundled apps then. I should have realised sooner to be fair. 
Have a great day!

jaylfc, 2023-10-04:
Just tried to compile 1.2.8 but got these errors (going to see if I can solve it)

In file included from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/main/main.cpp:1,
                 from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/M5Stack-SD-Menu.ino:80:
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h:66:19: note: #pragma message: M5Sack CoreS3 detected
   #pragma message "M5Sack CoreS3 detected"
                   ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h:149,
                 from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/main/main.cpp:1,
                 from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/M5Stack-SD-Menu.ino:80:
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/partition_manager.h: In function 'void lsPart()':
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/partition_manager.h:120:5: error: 'Flash' has not been declared
     Flash::digest_t digest;
     ^~~~~
In file included from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h:149,
                 from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/main/main.cpp:1,
                 from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/M5Stack-SD-Menu.ino:80:
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/partition_manager.h:126:15: error: 'Flash' has not been declared
       meta  = Flash::getSketchMeta( part );
               ^~~~~
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/partition_manager.h: In function 'void checkMenuStickyPartition(const char*)':
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/partition_manager.h:175:7: error: 'Flash' has not been declared
   if( Flash::getFactoryPartition() ) {
       ^~~~~
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/partition_manager.h:176:5: error: 'Flash' has not been declared
     Flash::loadFactory(); // should trigger a restart
     ^~~~~
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/partition_manager.h:190:10: error: 'Flash' has not been declared
     if (!Flash::comparePartition(running, nextupdate, sksize)) {
          ^~~~~
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/partition_manager.h:199:11: error: 'Flash' has not been declared
       if (Flash::copyPartition( flgSD ? &dst : NULL, nextupdate, running, sksize)) {
           ^~~~~
In file included from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/main/main.cpp:1,
                 from /Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/M5Stack-SD-Menu.ino:80:
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h: In function 'void drawM5Menu(bool)':
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h:588:19: error: 'flashUpdaterIcon16x16_jpg' was not declared in this scope
       tft.drawJpg(flashUpdaterIcon16x16_jpg, flashUpdaterIcon16x16_jpg_len, 8, 6, 16, 16);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h:588:19: note: suggested alternative: 'sdUpdaterIcon15x16_jpg'
       tft.drawJpg(flashUpdaterIcon16x16_jpg, flashUpdaterIcon16x16_jpg_len, 8, 6, 16, 16);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                   sdUpdaterIcon15x16_jpg
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h:588:46: error: 'flashUpdaterIcon16x16_jpg_len' was not declared in this scope
       tft.drawJpg(flashUpdaterIcon16x16_jpg, flashUpdaterIcon16x16_jpg_len, 8, 6, 16, 16);
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h:588:46: note: suggested alternative: 'sdUpdaterIcon15x16_jpg_len'
       tft.drawJpg(flashUpdaterIcon16x16_jpg, flashUpdaterIcon16x16_jpg_len, 8, 6, 16, 16);
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                              sdUpdaterIcon15x16_jpg_len
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h: In function 'void doFSChecks()':
/Users/jay/Downloads/M5Stack-SD-Updater-1.2.8/examples/M5Stack-SD-Menu/menu.h:964:23: error: 'Flash' has not been declared
   factory_partition = Flash::getFactoryPartition();
                       ^~~~~

exit status 1

Compilation error: 'Flash' has not been declared

tobozo, 2023-10-04:
if you don't know how to update an arduino library from a repository I suggest you wait for the next release to pop in the library manager

unless you want to contribute there's no need to use the 1.2.8 branch to create your SD-Menu binary or compile your SD-loadable applicationss

jaylfc, 2023-10-04:
Ok, I did update the library and those were the compile errors but never mind. Im actually attempting to get it working on this device for a project: https://github.com/Freenove/Freenove_Development_Kit_for_ESP32_S3
How feasible do you think this is, if needs be I'd happily pay for help :)

tobozo, 2023-10-04:
Freenove devkit does not seem to have a SD Card, I wouldn't recommend that board for a SD-Card based project :wink:

jaylfc, 2023-10-04:
It does indeed, built into the board.Sent from my iPhone

tobozo, 2023-10-05:
well good news then :+1: 

here's a roadmap to get M5Stack-SD-Menu to work on a Freenove devkit:

- establish the pin map (SD card, TFT, Touch, buttons)
- create a board profile (Arduino IDE/platformio)
- create an LGFX config profile
- create an ESP32-Chimera-Core device profile
- add Freenove devkit profile to M5Stack-SD-Menu
- test, test, test!

Freenove devkit is currently sold at ~65€ on [amazon](https://www.amazon.fr/Freenove-Development-ESP32-S3-Compatible-Wireless/dp/B0C2K34N7Q) in my country but I already spent my monthly budget, so if you want this in october you'll have to either send me a unit or paypal me so I can purchase it myself and start working on a proof of concept.

jaylfc, 2023-10-07:
Thanks for that, I'll give it a try. If I can't manage it I'll arrange for one to be shipped to you :)